SanaaHamel/nevermore-controller #9: Updating to v0.13 looses one set of sensors

mikeytdisco, 2024-03-29:
The nevermore display in Mainsail shows "-" for the left sensor values.  Flashing back to v0.12 shows the sensor values.

Dark360, 2024-03-30:
@mikeytdisco,

@SanaaHamel posted this on Discord to fix this issues whilst they worked on a solution.

# switch to your nevermore-controller installation
cd ~/nevermore-controller
# fetch updates for klipper module and tools
git pull
# reset the pin configuration to default. follow the on-screen instructions.
~/nevermore-controller/tools/pin-config.py --reset-default

[](https://github.com/SanaaHamel/nevermore-controller?tab=readme-ov-file#pin-assignments)

mikeytdisco, 2024-04-02:
The work around fixed the issue.  Thanks.

closed 2024-04-02 by mikeytdisco

SanaaHamel, 2024-04-02:
Should also be fixed by 0.14 release.