SanaaHamel/nevermore-controller #10: Update Manager indicating "Unknown"

stefanpieter, 2024-04-03:
After the latest Nevermore update via Mainsail, the update manager, the status is indicated as "Unknown." Once updated, Klipper was not able to connect to the MCU.

v0.14-1-ge58cbabb

I manually updated as per the [dummies guide](https://github.com/SanaaHamel/nevermore-controller?tab=readme-ov-file#updating-guide-for-dummies)

It now connects; however, the "Known" status remains in the update manager.

[image]

SanaaHamel, 2024-04-08:
> Once updated, Klipper was not able to connect to the MCU.
Insufficient information, would need printer log for details. Common cause: 2.4 GHz interference, see FAQ.

> It now connects; however, the "Known" status remains in the update manager.
Can reproduce. This is an issue with Mainsail.
Looking at the WS data coming from moonraker, it all looks sane. Reading the code suggests it might be upset with `semver.valid` rejecting `v0.14-1`, which isn't a valid semver according to the spec, but I also don't have any problems with my other tracked packages which do have equivalently 'malformed' version numbers according to the spec...
Maybe @meteyou  can provide insight?

stefanpieter, 2024-04-08:
As requested.

[klippy (23).log](https://github.com/SanaaHamel/nevermore-controller/files/14907911/klippy.23.log)
[moonraker (4).log](https://github.com/SanaaHamel/nevermore-controller/files/14907912/moonraker.4.log)

closed 2024-09-20 by SanaaHamel