SanaaHamel/nevermore-controller #15: Feature Request: Persistent Config File and Temperature Gating Threshold

7s3, 2024-08-22:
1. For those of us with a standalone version of the Nevermore Mini with a Waveshare touch screen (or do not have the ability to compile firmware ourselves), it would be useful to have a config file located at the root of the device that we can make changes to.

2. It would also be useful to have a new sensor setting for temperature.
Similar to `voc_gating_threshold`, a `temperature_gating_threshold` would be beneficial.
Very often my fan turns on when the printer is not even printing and the enclosure is at room temp due to sporadic VOC sensor readings. If we could set a temperature threshold it would make the carbon last longer.

Rooby69, 2024-10-24:
I have the same issue that I'm using it with Waveshare touch screen standalone. For me it would be helpful if I could connect via USB and virtual COM port and do some settings via terminal, which are then stored permanently on the device.

SanaaHamel, 2025-01-24:
As of v0.16, issue part 1) is resolved.