SanaaHamel/nevermore-controller #17: requirements.txt

O-to-the-o, 2024-09-08:
Hi the requirements.txt causes issues when i try to install it, it's fixed by chaning it to:

bleak>=0.21.1, <1
janus>=1, <2
typing_extensions>=4, <5

could anyone who knows more about github look at this pls ?

Kind regards
O

ksummers92, 2024-09-11:
Created PR here. Not sure how to link this issue though. Or link the PR here either.

eMadman, 2024-09-14:
Thanks @ksummers92 - I'm migrating a machine to a new pi and this solved the issue.

krompiel, 2024-09-19:
good morning gentleman. i ran into the same issue.

im terrible with linux, could i, and how would i correct this manually?

thank you

eMadman, 2024-09-19:
> good morning gentleman. i ran into the same issue.
> 
> im terrible with linux, could i, and how would i correct this manually?
> 
> thank you
Use a text editor like nano to change requirements.txt to match what's been attached to this pull request: https://github.com/SanaaHamel/nevermore-controller/pull/18

krompiel, 2024-09-19:
> > good morning gentleman. i ran into the same issue.
> > im terrible with linux, could i, and how would i correct this manually?
> > thank you
> 
> Use a text editor like nano to change requirements.txt to match what's been attached to this pull request: #18

well duh of course lol. stupid me. thank you sir i shall try this tonight

matt-rendall, 2024-09-20:
Thank you! Also ran into this when migrating to a new Pi.

PR #18 (2024-09-20): Update requirements.txt
Update requirements.txt to fix error during install. Fixes #17 .
closed 2024-09-20 by SanaaHamel