SanaaHamel/nevermore-controller #20: compatability issue with newsest klipper version

woodencliff, 2024-10-01:
Klipper v0.12.0-315-g8f361a15 updates some led related function, and after update, an error called "Unable to load module 'led'" pop up. According to log, it is caused by nevermore.py. Maybe there is some interference between the new klipper led code and nevermore-controller code?
[klippy (15).log](https://github.com/user-attachments/files/17199573/klippy.15.log)

Squid116, 2024-10-01:
I'm also hitting this - but don't utilize any LEDs on my nevermore.

Squid116, 2024-10-01:
FWIW resetting klipper to the version from before this change allowed me to avoid the error (for now):
```
cd ~/klipper
git reset ef75346861fdf94d952a4e887fb843d215b00631 --hard
```

SanaaHamel, 2024-10-04:
Tentatively fixed by 4c69464a0a83d4afba1fc8a28567378ab0b75ed2...

closed 2024-11-15 by SanaaHamel