SanaaHamel/nevermore-controller #21: Klipper Fails to Reconnect After Emergency Stop

padok, 2024-11-22:
Problem:
After initiating an EMERGENCY STOP, Klipper is unable to successfully reconnect to the Nevermore controller. This results in the printer being inoperable until a powercycle of the nevermore controller is performed.

Expected Behavior:
Following an EMERGENCY STOP, Klipper should be able to automatically reconnect to the Nevermore controller.

Steps to Reproduce:
1. Initiate an EMERGENCY STOP
2. Restart Firmware
3. Klipper not starting due to missing nevermore controller connection

Klipper Version: v0.12.0-377-g9bd0d475 
Nevermore Controller Version: v0.15.1-3-g4c69464a

padok, 2024-11-26:
I may also add, that the controller seams to lock up completely sometimes. After a just finished print I'm unable to change the fan speed and the sensor data also does not seam to be updated anymore.

Is there a way to force reset the controller from klipper?

padok, 2024-11-26:
The update Ota script also does not find the controller anymore in this state. 

`nevermore-controller/tools $ ./update_ota.py
Tool environment seems up to date.
Klipper service is active.
Due to software quirks you should stop klipper during the firmware update.
(Make sure you don't have a print in progress!)
Stop Klipper during update? [Yn]
Stopping klipper...
RELEASE TAG: v0.15.1
RELEASE DESCRIPTION:
Fix `voc_improve_min` policy activating when intake < exhaust.

**Full Changelog**: https://github.com/SanaaHamel/nevermore-controller/compare/v0.15...v0.
15.1

PRESS ENTER TO CONTINUE
discovering Nevermore controllers...
WARNING:root:no Nevermore controllers found
attempting to connect to bootloader anyways...
Unable to connect to nevermore to query board. Please specify board.
Board:`

sntlewis, 2024-11-29:
I have also experienced both issues.  (posting to bump the issue).

SanaaHamel, 2024-11-30:
> I may also add, that the controller seams to lock up completely sometimes. After a just finished print I'm unable to change the fan speed and the sensor data also does not seam to be updated anymore.
> 
> Is there a way to force reset the controller from klipper?

This usually a lost connection rather than a controller hang. Unfortunately usually that's on the central device's (the SBC) side rather than the controller. You can verify this by connecting to the controller using another device (e.g. your phone & nRF Connect).