SanaaHamel/nevermore-controller #26: Hardware support request: Waveshare ESP32 1.28 Touch Screen

Jo0, 2025-03-19:
Noticed today that the [Waveshare ESP32 1.28 Touch](https://www.waveshare.com/esp32-s3-touch-lcd-1.28.htm) exists. Are there plans on supporting this unit to overcome the current lack of BT LE on the current RP2040 1.28 Touch model? I noticed on other posts there are mentions of requests being declined without sponsoring, would that be like a donation through buymeacoffee or hardware donation?

SanaaHamel, 2025-03-19:
Politely: Not going to happen unless you're a vendor and you're willing to sponsor the port for a 2k+.

Porting to ESP32 would be a half-rewrite of the codebase and a new bootloader.

closed 2025-03-19 by SanaaHamel

SanaaHamel, 2025-03-19:
The rust `cyw43` crate claims to implement support for pico w devices, and embassy-rs/trouble claims to offer classic/BLE host side support. This has gotten more plausible, but issues remain w/ platform specific details re' the bootloader and persistence store.