SanaaHamel/nevermore-controller #29: Default value causes json parsing error

keyz182, 2025-08-27:
https://github.com/SanaaHamel/nevermore-controller/blame/3cf62a1b9fcd17e000086cfbdebb4ceb3b2c84cb/klipper/nevermore.py#L841

When this takes the default value, the json output from klippy becomes the following, which is not valid json. 
```
...
"print_mode_min_bed_target":-Infinity
...
```
 

This causes moonraker to fail to start with 
```
Traceback (most recent call last):
  File "/home/pi/moonraker/moonraker/components/klippy_connection.py", line 206, in _read_stream
    decoded_cmd = jsonw.loads(data[:-1])
                  ^^^^^^^^^^^^^^^^^^^^^^
msgspec.DecodeError: JSON is malformed: invalid character (byte 157948)
```

ImSundee, 2025-09-04:
Same issue here, setting to -9999999 resolved the issue.

SanaaHamel, 2025-09-16:
Resolved by [36013e6](https://github.com/SanaaHamel/nevermore-controller/commit/36013e6540a0690e0f34a0bcca41848c8f48dd2a)

closed 2025-09-16 by SanaaHamel